iNavFlight/inav-configurator #726: [SOLVED]  INAV Failed to open serial port on WINDOWS

aletev, 2019-03-18:
Hi everybody, for me was very hard to solve this "problem". Wasn't easy to find something about it. And as I shared it with a group of friends and they told me that for them was a problem too, I decided to write this lines.
I could solved it just for windows, not for ubuntu. (for the moment).
The key was [https://impulserc.com/pages/downloads](url).
Just this steps:
1) Download ImpulseRC Driver Fixer
2) Connect the FC to the usb.
3) Run ImpulseRC Driver Fixer. And wait until it connects to the FC and retrieves drivers info an update the loaded ones.
4) Restart... and now you can flash your FC. 
In my case is an OMNIBUS SD F4 V5. Flashed with OMNIBUSF4V3 (inav).

Bye.

closed 2019-03-24 by DzikuVx

julienguegan, 2020-07-14:
I also have "Failed to open serial port" and try to use ImpulseRC driver but it gives me an error ... 
[image] 

If someone has a solution for this, it would be great !

DzikuVx, 2020-07-14:
@julienguegan probably a different program is blocking COM port

julienguegan, 2020-07-21:
@DzikuVx Did you already encountered this error ? I can't figure out if there is something blocking the COM port ...

I also try to launch zadig but in devices list I don't have the "STM32 Virtual COM port"
[image]

shenshihehe, 2024-06-23:
nice suggesition