iNavFlight/inav-configurator #729: Motor 1 (ESC #1) not spinning up, every other motor works fine

cnsumner, 2019-03-21:
Not sure if this belong here or in the firmware issue tracker, correct me if I chose wrongly.  

I have a betaflight F4 FC that I am wanting to flash iNav onto for gps related things.  Flashing went fine, everything seemed fine, but when going to test the motors, motor #1 does not respond. It plays it's little startup diddy, but will not spin up, every other motor works.  I have tried every single ESC protocol, even going back and reflashing betaflight to verify that each protocol does indeed work with my ESCs.  Going into blheli configurator shows the selected protocol for each ESC, except #1 shows "unknown".  I'm using crazepony blheli 32 35A ESCs, and verified that they do indeed work with oneshot, since dshot won't save on iNav (reverts to standard).  Any ideas on why only one ESC wouldn't spin up?

shellixyz, 2019-03-21:
It should have been posted in the firmware tracker but no worries. It is a known bug. The first output definition haven't been ported correctly from 2.0.1 to 2.1. Try this build it should fix your issue: [inav_2.1.0_BETAFLIGHTF4.hex.zip](https://github.com/iNavFlight/inav/files/2923615/inav_2.1.0_BETAFLIGHTF4.hex.zip)

cnsumner, 2019-03-22:
Worked like a charm. Thank you!

closed 2019-03-22 by cnsumner