iNavFlight/inav-configurator #730: Global Mixer functions

Jetrell, 2019-03-24:
I have picked up that @DzikuVx is working on some form of Global Mixers for iNav.
But I have not found much information on what these mixers will be able to do?

I have a suggestion Pawel, for a particular mixer, that could be used for OSD page conditional mixing.

For example. At present I use Lua Telemetry to tell my Taranis when things like RSSI is low, so I can mixer it to the OSD page selection channel. 
It will then switch back to the first OSD page that has my GPS coordinates on it, at less than 30% signal strength.
At present the only function that does this in iNav, will only do so when a Failsafe occurs. But by this time the video feed may have also been lost.

The above example works well within telemetry range. While a Global mixer that works the same way would be more ideal for long range fixed wing flights.

There are a large number of possibility that a Global mixer can be used for when it comes to OSD page and information selection. If you like cruising around without the OSD information being displayed on your live feed.

Let me know what you think guys?

DzikuVx, 2019-03-24:
Well, another good options for global functions. It's not yet documented since it's not yet done. When general framework will be in place, we can think about things that this thing can do

noagendafpv, 2019-05-30:
> Well, another good options for global functions. It's not yet documented since it's not yet done. When general framework will be in place, we can think about things that this thing can do
Is there a plan to allow Logic Conditions to be applied to the Motor Mixer? Specifically to each of the Motor lower level tie ins, Roll/Pitch/Yaw? This would allow for say yaw differential motors to be activated or deactivated, instead of always off or on.

MrD-RC, 2022-11-26:
@Jetrell does the current programming framework fulfil your request?

Jetrell, 2022-11-26:
Yes it does.

closed 2022-11-26 by MrD-RC