iNavFlight/inav-configurator #734: Failure during reboot - Stuck in DFU mode

Lore88, 2019-03-24:
Hello,

I'm working on a fpv s800 wing with a MATEK f411 wing and i have an issue when i plug it on inav. Every time i try to save a change and the board requires a reboot, the board reboots and goes into DFU mode.
Sometimes i can get back into INAV just by unplugging and plugging it back into the usb but some other times it keeps going straight into DFU mode and the only way to get back into inav is to flash it again with new firmware or to leave it sit for a while and then plugging it back in.

I though the issue may have been releted to drivers somehow but even after updating, repairing and removing stm32/dfu drivers the issure remains.

I olso had times where the board would work fine for a while but then all of a sudden, always after rebooting to save some changes, it would go straight back into DFU (and no, its not an issue with the boot button :) )

Any ideas of how to fix it ?

Thanks,
Lorenzo

shellixyz, 2019-03-24:
The issue is most likely your DFU/BOOT button getting stuck. Try hitting it with some electrical contact cleaner and if it still doesn't work replace it.

shellixyz, 2019-03-24:
Or maybe a solder ball or something else conductive on your board shorting the boot pad of the MCU. Anyway it most likely doesn't have anything to do with iNav/drivers

closed 2019-03-24 by shellixyz

Lore88, 2019-03-24:
> The issue is most likely your DFU/BOOT button getting stuck. Try hitting it with some electrical contact cleaner and if it still doesn't work replace it.

thanks for the quick reply.
I'll try and look into that a bit more in details.
I wouldn't wanna remove the button and mess the board up as it is brand new and i'd rather try and get it replaced by the seller.
Thanks for now,
Lorenzo