iNavFlight/inav-configurator #735: Can't connect OMNIBUSF4PRO on Mac

flybabo, 2019-03-26:
All of a sudden, I can't connect OMNIBUSF4PRO from inav-configurator 2.1.4 on MacBook Pro after flashing it with the iNav 2.1 firmware downloaded from online. (I can put this FC to DFU mode and flash it)
inav-configurator complained that "No configuration received within 10 seconds, communication failed".  I've never had such a problem before.
I tried the Betaflight firmware in the same manner and the betaflight configurator connects to this FC on the same MacBook Pro.
Any clues?

NOTE: This problem is also reported as inav/issues/4547 since I'm not sure which side caused this problem.

shellixyz, 2019-03-30:
Could you please try with Windows or Linux ?

DrJ-FPV, 2019-04-06:
Which brand and is it a V2 or a true V3? ELSE: https://github.com/iNavFlight/inav/wiki/Supported-boards

[image]

flybabo, 2019-04-06:
Unfortunately, I don't have any real Linux/Windows platforms for testing - they run on the virtual machine on my MacBook Pro.
My Omnibus F4 Pro is a V3 and I don't think this problem is related to the Club-24A problem.
(Never heard about the Club-64A - 64Amps flow thru the current sensor?)
Anyway, I got a new Omnibus F4 Pro V3 and current iNav combination works fine with the new one.
I still don't know what really causes this problem - the old one had worked fine until recently and it still works fine with the Betaflight firmware.  A subtle timing issue in SPI bus transaction may lock up the FC as someone commented on another thread.
I'm closing this issue.

closed 2019-04-06 by flybabo

lumenier1, 2020-12-03:
I've also ordered fly tower (F4 Matek411 target) from TCMM but instead of the original FC a clone was delivered.
There is the same issue on MacBook: "No configuration received within 10 seconds, communication failed".
Also I've tried it on Windows, so FC was able to connect just for 1 time (without gyro/accelerometer response), and then the same issue again as for Mac.
Then I've tried BetaFligft on Mac, it connects but without any data for the gyro/accelerometer, FC frezed on one position at zero angle in space.
So, it was definitely damaged FC with USB, i2c or UART chip issues or other replica SMD components.
Buy only original ones.