iNavFlight/inav-configurator #738: Baro sensor not detected with LM75 I2C connection

Jetrell, 2019-03-31:
I have an LM75A temperature sensor breakout board connected to an F411-Wing flight controller.

[image]

Prior to the connection of the LM75A. I could connect the F411-Wing to the configurator and there would be NO hardware errors displayed. And the BARO sensor indicator would be blue.
Now, when I have the LM75A connected to the I2C bus (barometer is also by default). The configurators  BARO indicator is red and a hardware error appears.

The barometer and temperature sensor seem to function okay in ground tests, according to the OSD values. But I am not willing to take it for a flight until I understand why this is happening. In case the I2C bus fails to see the baro sensor in flight.

@shellixyz  Because you are the temp sensor author, can you see any reason that this should be happening?
Thanks

shellixyz, 2019-03-31:
I don't know what is going on. It should work just fine. I'm going to test with a Matek F411-Wing and a LM75A to see if I can reproduce the issue

shellixyz, 2019-03-31:
@digitalentity Could you please move this to the firmware tracker

Jetrell, 2019-03-31:
Thanks for the reply @shellixyz 
I found what was causing it.
Because this boards I2C pinout does not have the GND inline with the 3.3v/data/clock pins. 
I used the 5v regulator instead of the 3.3v regulator to drive the LM75.
My reason was because the board in mounted up side down. And I prefer to use at least a 2pin dupont header for mechanical strenght over a single pin. (having the VS+ and GND together on the one header)

It appears that the 5v 2amp regulator and the 3.3v LDO must be fighting each out through I2C bus....some how.
Does that sound like a possibility to you?

shellixyz, 2019-03-31:
If your LM75A module has some on-board pull-up resistors it could explain the issue. Luckily you don't risk damaging the F411 MCU since the I²C pins are 5V tolerant. Not sure about the BMP280. But if it still works everything should be ok.

Jetrell, 2019-04-01:
Thank you @shellixyz  for your insight.
I think i'll have to just wire it up to the 3.3v regulator for piece of mind.

Can I ask a question and even make a feature request?
Is there any reason that you didn't include the option to use a 10k NTC thermistor for a temperature sensor?
I do understand that it would require some form of calibration and that it is not as actuate as a digital device. 
Plus there are not many ADC pins on most FC's.

 The reason I would still like to request this option. Is because there is not a lot of room in the small planes that we install the F411-Wing FC into.
And in these small wings we generally use small electronic speed controllers. Which leaves the LM75 breakout board being almost as large as the ESC, which blocks its cooling efficiency.
While a thermistor is very small and can be slipped under the heat shrink tube of the ESC, with ease.

Plus. someone also made a firmware for an ADC, to work with the F411-Wing ;-) 
I use RSSI/LQ on a radio channel. So the ADC firmware would be ideal for a thermistor. Plus it would only use 2 wires instead of four. In these small applications.
[image]

I have used DS18b20's with a DS2482, for a larger build, when multiply sensors were used. 
While with a smaller build, the DS18b20 is small in size, but the DS2482 makes it a bit impractical.

Sorry for the long speech. I'm sure you get the idea.  

Thanks

shellixyz, 2019-04-01:
We can add the possibility of using a thermistor it is true that it would be an easy way to connect a small temperature sensor. 

To connect the DS2482 and LM75A chips I'm using small breakout boards that are approximately 11x13mm: https://www.ebay.com/itm/100pcs-SOP8-SO8-SOIC8-TO-DIP8-adapter-pcb-conveter-board/201418796668

I will probably also add an 1-Wire UART driver in the future to allow using an UART instead of a DS2482 chip on I²C to connect DS18B20 sensors.

Please close this if the issue if resolved and open a feature request for the thermistor support.

closed 2019-04-01 by Jetrell