iNavFlight/inav-configurator #739: Request for Temp sensor addition (Thermistor)

Jetrell, 2019-04-01:
I would like to request this option. To keep the weight and size down in smaller models. That also use smaller devices like ESC's.
In the case of an LM75 breakout board. It is almost as large as the ESC, which blocks its cooling efficiency.
While a thermistor is very small and can be slipped under the heat shrink tube of the ESC, with ease.

The RSSI ADC pin on most Flight Controllers that is not used by many of us. Who put RSSI/LQ on a radio channel. Could be the option for this.
And it only uses 2 wires instead of four. To simplify the application.

[image]

shellixyz, 2019-04-01:
@digitalentity Could you please move this issue to the firmware repos

Jetrell, 2019-04-02:
Sorry about that @shellixyz 
I was not sure which to place the request under, cause it is see in both Configurator and firmware.
I'll close it here and re-open it there.

closed 2019-04-02 by Jetrell