iNavFlight/inav-configurator #740: Program error iNAV since 2.0, a coup in the air at 100% of the throttle

Nikxxxxx, 2019-04-02:
I beg your help! This problem is not only bothering me alone. At a sharp acceleration, the level of the choke damper 100%, the drone makes a random Somersaults in the air. This problem is observed with firmware 2.0 and above. For comparison, added a log from the Betaflyght, with the same drone and there are no such problems. I beg your help)
[Inav + betaflight.zip](https://github.com/iNavFlight/inav-configurator/files/3033406/Inav.%2B.betaflight.zip)

Jetrell, 2019-04-03:
I agree. I've had a similar experience above 93% throttle, even with AIR mode enabled. 
I'm running 2.1.
I was thinking of loading Betaflight to see if it reacted the same.
After looking over my logs and DVR footage. I put the cause down to the motors not having anymore power to give for attitude/stabilization control, when the throttle is above 93%. 
 Mine tends to over compensate, when the throttle is reduced in a turn. And will flip inwards up to 180°, before it will stabilize.  
Its very UN-nerving when traveling at high speed in a tight area.

Nikxxxxx, 2019-04-03:
> Согласен. У меня был подобный опыт выше 93% газа, даже с включенным режимом AIR. 
> Я бегу 2.1. 
> Я думал о загрузке Betaflight, чтобы увидеть, реагирует ли он так же. 
> После просмотра моих журналов и видеозаписей DVR. Я объяснил причину тому, что двигатели, у которых больше нет мощности для управления положением / стабилизацией, когда дроссельная заслонка превышает 93%. 
> Шахта имеет тенденцию к чрезмерной компенсации, когда дроссельная заслонка уменьшается в повороте. И сместится внутрь до 180 °, прежде чем стабилизируется. 
> Это очень нервно, когда едешь на большой скорости в стесненных условиях.

I tried with different firmware 2.0 and ending 2.1. In, flew in different flight modes. The Problem persists everywhere. Did You somehow solve this problem? You say motors do not go to the right level, why then in betafly everything works fine

Jetrell, 2019-04-03:
I did not solve the problem in iNav. And I never had the same issue with an F1 board in Cleanflight using the same old air frame, Which now has an F4 board and iNav.
I think the problem may not be seen the same in Betaflight. Because Betaflight is more designed around performance. While iNav is more designed around sport flying with GPS navigation.

I would hope that the dev's may be able to fix this some what. But iNav has a far more tasks running in the background of its code, than Betaflight does.  So I would imaging that the PID loop time in iNav would suffer to a degree.  Which is what I think may be contributing to the effect.

It is still good that you brought this issue up. At least they are aware that it exists.

Nikxxxxx, 2019-04-03:
> Я не решил проблему в iNav. И у меня никогда не было той же проблемы с платой F1 в Cleanflight, использующей ту же старую воздушную раму, которая теперь имеет плату F4 и iNav. 
> Я думаю, что проблема не может быть замечена в Betaflight. Потому что Betaflight больше ориентирован на производительность. В то время как iNav больше ориентирован на спортивные полеты с GPS-навигацией.
> 
> Я надеюсь, что разработчики смогут это исправить. Но в фоновом коде iNav гораздо больше задач, чем в Betaflight. Поэтому я хотел бы представить, что время цикла PID в iNav пострадает до некоторой степени. Что, я думаю, может способствовать эффекту.
> 
> Все еще хорошо, что вы подняли этот вопрос. По крайней мере, они знают, что это существует.

Yes, the problem really makes the situation uncomfortable in flight. And This is not only on my Matek f405 Ctr, my friend has the same problem, the same Matek f405ctr, and another flight controller on the F7 omnibus. 

I hope the developers will pay attention to mine and not only my problem)

DzikuVx, 2019-05-24:
this is FC tuning not configuration tool issue

closed 2019-05-24 by DzikuVx