iNavFlight/inav-configurator #741: Win64_2.1.4 iNav on win10 gfx compatibility issue

DrJ-FPV, 2019-04-04:
On Win10 1803 (OS build 17134.648), 64 bit, iNav_2.1.4 loads to the taskbar but is a no show regarding the iNav 'start-page'. Win TAB switching shows an iNav window running but unable to switch into active iNav window.

iNav Chrome app works fine, although the mission planner is on the 2.1.4 9 (dot)exe so that remains unaccessible for now.

Requesting remedy, most pc's today are sold with win10-64 pre-installed.

Thanks! Keep up the good work.
DrJ-FPV

[image]

[image]

shellixyz, 2019-04-05:
Try to delete the configuration folder:
C:\Users\%Your_UserName%\AppData\Local\inav-configurator

DrJ-FPV, 2019-04-06:
Solid answer. Thanks! 

Applied 'more heads than one' theory and repeated task, no changes observed. OS and drivers(stable) are up to date on the machine, (dot)exe is not blocked anywhere either.

-Unresolved.- Any more ideas?

shellixyz, 2019-04-06:
I have no more ideas. Seems like has something to do with your specific Windows install since I ran it on Win10-64 without any issue (Intel GFX driver). We can hardly fix an issue we can't reproduce.

DrJ-FPV, 2019-04-06:
Thanks for your response. 

While I agree that you cant fix a problem we cannot reproduce, I do not when it comes to your assessment of the situation as a whole. Seems lazy to blame it on a specific install and then call it a day. Thats level 1 or 2 support stuff, no offense. I was really just hoping to use a scalpel, not a hammer..

igfxEM process listed above indicates Intel gfx, yup.

Checked local HBSS software, no blocks.
Deleted, refreshed Configuration folders. Nothing.
Downloaded a new 2.1.4 zip, refreshed Configuration folders. Nothing.
Dabbled in safe mode with above options. Nothing.
Checked the registry, couldnt find any specific entries.
Tried various compatibility modes. Nothing.

-ISSUE RESOLVED-
Changed program DPI, changed DPI scaling behavior of inav-configurator. Apparently iNav as an application cant perform high DPI scaling behavior on its own, changed "performed by" to "System".
Works.

[image]

DrJ-FPV, 2019-04-06:
Closing thread, issue resolved. Thank you to all involved.

closed 2019-04-06 by DrJ-FPV