iNavFlight/inav-configurator #742: Motor Tab - Offset on the servos

lichtl, 2019-04-06:
Hi,

it seems that on the motor page the servo count is off.

[image]
[image]

```
# diff

# version
# INAV/FF_F35_LIGHTNING 2.1.0 Feb 25 2019 / 16:46:34 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 3 1000 2000 1634 -100
servo 4 1000 2000 1469 100
```

shellixyz, 2019-04-06:
Well yes and no. Servo indexes start at 0. So servo 3 in the mixer is servo 4 in the servos tab (if servo index 0 is the first servo, servo index 3 is the fourth servo). Not the most straightforward I agree. Just explaining how it is.

lichtl, 2019-04-07:
yes thought so as well. however would be probably better to have them consistent especially when the model is shown in top left which then result in more risk of confusion.

DzikuVx, 2019-07-16:
Changed in #806

closed 2019-07-16 by DzikuVx