iNavFlight/inav-configurator #746: (New) Impact mini font

Jetrell, 2019-04-16:
This font is modeled off the Impact font. But it has a smaller stature in all characters. To take up less OSD screen space. But is still easy to read at a glance. 
 
[image]

[Impact_mini.zip](https://github.com/iNavFlight/inav-configurator/files/3082487/Impact_mini.zip)
[Impact_mini .mcm.zip](https://github.com/iNavFlight/inav-configurator/files/3082488/Impact_mini.mcm.zip)

If any of the Dev's think it is worthy of a PR and to merge. Then feel free to do so.

wx4cb, 2019-05-04:
i like it, but maybe make the stroke thinner a little some of the numbers seem a little "cramped"

Jetrell, 2019-05-04:
@wx4cb Its only two pixels wide. If it goes to one pixel....it becomes like Default font.

The idea was to keep it FULL in its look, but still be a small in stature.  So that's the best it can be, with the available pixel density  :-) 
I find it looks better in the goggles than the preview here.

wx4cb, 2019-05-05:
@Jetrell ok cool... hard to tell from the preview image

PR #877 (2020-01-25): Add Impact mini OSD font
Closes #746 

Thank you for your contribution @Jetrell
closed 2019-12-03 by Jetrell