iNavFlight/inav-configurator #748: OSD font don't upload Inav 2.1.0 F722

spacyje, 2019-04-21:
Hi,
I installed the latest version of inav 2.1.0 on a matek F722 card and I can't upload the font, it still blocks about half of the font upload, I did some research and I went to the configuration page to enable/disable the osd with lipo, without lipo but it's always the same.
On the 2.0 I didn't have this problem.
Any solution?

jimokeane, 2019-04-28:
I also have this problem. I tried every possible thing I could think of based on other suggestions. OSD switch on/off. Making sure board had battery power, .etc. I found that no matter what the upload of the font stopped about at 300 / 500 or something like that. Font was then corrupted in the OSD! I went back to the previous version of the configurator and uploaded the old font. This upload works fine, but when you boot the F722 Wing, it shows a message that the font is invalid.

jhyden, 2019-04-28:
I had the same problem with a f722se and a Mac. Went back to 2.0 and the upload works.

shellixyz, 2019-04-28:
@spacyje @jimokeane Are you using MacOS ?

jimokeane, 2019-04-28:
Yes - I am running MacOS 10.13.6 (17G6030)

JohnDevine, 2019-05-16:
Just commenting .. same here Mac os x, F722-se font load stalls at 304/512

JohnDevine, 2019-07-22:
> Just commenting .. same here Mac os x, F722-se font load stalls at 304/512

Still in 2.2.0

Flyboy369, 2019-07-22:
Same for me.  New to FPV, with iNav 2.2.0 font upload failed at 257/512 every time on my mac with matek F722-SE  Downloaded 2.0.0 and it worked great although when the fc boots up the OSD screen states "Invalid Font"

fiam, 2019-07-27:
Just tested with the PR in #800, which fixes the problems on F3 with macOS, and it turns out it also fixes the problem for F722. We'll be shortly releasing configurator 2.2.1 with this change.

DzikuVx, 2019-07-27:
2.2.1 released so we can close this. Thank you very much

closed 2019-07-27 by DzikuVx