iNavFlight/inav-configurator #749: When the map isn't available due to no API or other reason, display an error on the screen

laurentopia, 2019-04-22:
instead of leaving the screen gray
people get confused, don't know why the map isn't showing up
[image]
Ideally: add to the error message what to do: click on gear and select the free map provider.

rholt51, 2020-03-11:
"click on gear"? "select free map provider"? is this a Mac (apple) thing one can do?
i'm using Windows 10 and lost my map a coupla months ago. i've been trolling for a solution.
i posted this on a site and it's still up to date:
20200209.... WIndows 10, INav v2.4.0
Matek - F722-SE, Beitian BN-880
8+ satellites.
no MAP on GPS just the spotter icon that moves around when i zoom in .
so maybe, in Windows 10, or maybe Google?, how do you "**change your map preference**"????
Russ from Ocala, Florida, USA

ZvEroBoY36, 2020-06-13:
> вместо того, чтобы покидать экран, серые
> люди запутываются, не знают, почему карта не отображается в идеале: добавьте к сообщению об ошибке, что делать: нажмите на шестерню и выберите бесплатный поставщик карт.
> [image]

А по точнее можно сказать куда и где найти эту шестеренку ? И можно скрин сделать по ближе что бы можно было увидеть что и куда нажимать ? Нету шестерёнки.

ZvEroBoY36, 2020-06-13:
У меня такая же проблема, как её можно решить ?

ZvEroBoY36, 2020-06-13:
> "нажмите на передачу"? "выбрать поставщика бесплатной карты"? это Mac (яблоко), что можно сделать?
> Я использую Windows 10 и потерял свою карту пару месяцев назад. Я пытался найти решение.
> я разместил это на сайте, и он все еще
> актуален : 20200209 .... Windows 10, INav
> v2.4.0 Matek - F722-SE, Beitian BN-880
> 8+.
> нет карты на GPS, только значок наблюдателя, который перемещается при увеличении.
> так что, может быть, в Windows 10 или, может быть, в Google ?, как вы « **меняете настройки карты** » ????
> я читал в одном или двух местах об «API»?
> Каким бы ни было исправление для пользователя Windows 10, может кто-нибудь дать инструкции о том, как вернуть карту?
> заранее спасибо,

Брат,карта больше не работает с Хром, установи отдельную програму айнав конфигуратор и будет тебе счастье

closed 2025-01-19 by b14ckyy