iNavFlight/inav-configurator #750: "Saving dataflash to file" dialog window does not close after downloadin bb data(Mac)

Troynica, 2019-04-24:
After or at the end of the downloading of the blackbox data (via a UART) from the onboard flash chip, the "Saving dataflash to file" dialog window shows the progress bar almost at the end but then it "hangs". By "hanging" I mean that the progress bar never reaches the end, nor does the dialog respond to clicking on the "cancel" button. It does, however, close when hitting the escape key on the keyboard.
The saved files appears to be complete, possibly just missing a few datapoints at the end, but I cannot tell and I can certainly live with it.
Because the black box data appears to be all there, for me the issue is minor/cosmetic (at least from the moment I found out pressing esc closes the dialog).
Filing a bug just to let you know. Perhaps there is a simple solution.

I only tried this with an old Quanton F4 board.

Troynica, 2019-04-24:
When I comment out:
```
        if (CONFIG.apiVersion && semver.gte(CONFIG.apiVersion, "2.0.0")) {
            buffer.push(lowByte(4096));
            buffer.push(highByte(4096));
        }
```
in script.js (line 11466) so that payload size is reduced (to 256?), the issue is gone. So it is possible that something is waiting indefenitely for a complete block. That will happen when the black box data size is not a multiple of 4096.

DzikuVx, 2019-09-23:
Configuratior 2.2.1 probably fixed this issue. @Troynica could you please verify?

closed 2019-09-23 by DzikuVx