iNavFlight/inav-configurator #760: ubuntu/Linux unable to access save/reboot buttons

wx4cb, 2019-05-04:
under any form of linux (mainly using ubuntu 18.06) whenever I try to get to the buttons they don't show as shown in this picture. the window is maximised etc but there's no way to get at them

my screen resolution is shown in the xwininfo shell. This is bviously a resolution issue with the configurator not respecting WM sizes.

wx4cb, 2019-05-04:
https://imgur.com/ZrmPHO5

shellixyz, 2019-05-16:
Looks like there is a minimum size for the window. Until this is fixed you should be able to move the window up to reach the save button by holding the super key, clicking anywhere on the window and dragging it up

wx4cb, 2019-05-16:
i tried that but it wouldn't work on the smaller screen.

DzikuVx, 2020-03-03:
#942 allows for 800x600 window size

closed 2020-03-03 by DzikuVx