iNavFlight/inav-configurator #784: How does the configurator flash firmware without connecting the connector

liutairan, 2019-05-29:
I have a question that how does the configurator flash firmware without connecting the connectors? 

I am trying to build a simple configurator program (with Python) to flash the firmware to SP Racing F3 boards. I found that when I use iNav configurator, I only need to connect the board via USB, then I can flash the firmware easily. But if I use another software "stm32flash" (https://sourceforge.net/p/stm32flash/wiki/Home/), then I need to connect the boot connectors.

So, what are the tips you use that can do the flash without connecting them? Thank you!

shellixyz, 2019-05-30:
You can send `R` through the serial port to reboot into DFU (bootloader) then use dfu-util to flash a firmware

liutairan, 2019-05-30:
> You can send `R` through the serial port to reboot into DFU (bootloader) then use dfu-util to flash a firmware

Thanks for the reply. What do you mean send `R` through the serial port?

liutairan, 2019-05-30:
After a lot of trial and error, I finally had dfu-util installed. However, it cannot find my board while using `ls /dev/tty*` the device can be found as `/dev/tty.SLAB_USBtoUART`.

liutairan, 2019-05-30:
I probably know how to do this now. Thanks.

closed 2019-05-30 by liutairan