iNavFlight/inav-configurator #792: Compass calibration on version 2.2.0-RC1 and 2.2.0-RC2 (OS-X)

Troynica, 2019-06-10:
When calibrating the compass, after the 30 seconds, no values for X, Y and  are shown. After saving and rebooting, all are still 0. Manually changed values are also not saved to the FC.
Calibrating does work with the 2.1 version of the configurator, but I don't know if these values can (safely) be used in 2.2 version of iNav.

Troynica, 2019-06-10:
Additionally, when a negative number is entered manually in any of the three fields, it is replaced by "0" when selecting a next field.

KenImhof, 2019-06-11:
I am having the same issue.  I am running OMNIBUSF4V3 2.1.  Earlier today, the configurator prompted to upgrade to the 2.2.0 version.  Now I am unable to complete the MAG calibration.  After save and reboot, all MAG values are still zero and flight controller status shows MAG not calibrated.  As as work-around, I connected to the Speedy Bee app on my phone to complete the calibration.

stronnag, 2019-06-15:
You can use values from a prior version, via the CLI if necessary.

Troynica, 2019-06-16:
Good to know. However, as someone mentioned in the forum, if it's a changed configuration (or a complete new build) recalibration is most certainly needed. I can still calibrate the compass by stick command. Regardless of how the values are obtained, using "get magzero" in the configurator CLI will show me the saved values. If needed, "set magzero_[xyz] = <value>" will rewrite these to the FC.

stronnag, 2019-06-16:
> Good to know. However, as someone mentioned in the forum, if it's a changed configuration (or a complete new build) recalibration is most certainly needed.

Not in my experience, a device calibrated some 3 years ago on a quad, now on a tricopter,  via numerous component upgrades works perfectly with the original calibration. But it's your choice; I prefer to avoid necessary inconvenience when a device is obviously working correctly.

Quadcopterman1948, 2020-03-25:
I currently have two new builds and iNAV Version 2.2.0 will not allow both compasses to be calibrated. As a result, the quadcopters cannot be armed. However, if I revert back to iNAV Version 2.1.4 both compasses calibrate successfully. Is it possible that this issue could be fixed by the iNAV Team in the next release? In the meantime, any suggestions from any of you experts out there would be a great help.

DzikuVx, 2020-05-12:
fixed by #992

closed 2020-05-12 by DzikuVx