iNavFlight/inav-configurator #801: Weird values for custom motor mixer in configurator window.

rrk1, 2019-07-02:
Inav 2.1, configurator 2.1.4

I am entering custom mixer for asymmetrical hex frame:

mmix reset

mmix 0  1.000 -0.420  1.000  0.900
mmix 1  1.000 -0.580 -1.100  0.890
mmix 2  1.000  0.420  1.000 -0.900
mmix 3  1.000  0.580 -1.100 -0.890
mmix 4  1.000 -1.000  0.000 -1.000
mmix 5  1.000  1.000  0.000  1.000

okay, checking with dump

# dump

# version
# INAV/MATEKF405 2.1.0 Feb 25 2019 / 16:58:21 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

mmix 0  1.000 -0.420  1.000  0.900
mmix 1  1.000 -0.580 -1.100  0.890
mmix 2  1.000  0.420  1.000 -0.900
mmix 3  1.000  0.580 -1.100 -0.890
mmix 4  1.000 -1.000  0.000 -1.000
mmix 5  1.000  1.000  0.000  1.000

Well, again looks good. Worked that way for years, back from BF2.6

Now checking what is going on in configurator, mixer window.

We are getting -0.421 and 0.41999 instead of 0.42 for example. Why? Rounding errors? Correcting values in configurator would not help, going further with both errors in mixer screen and cli dump now.

rrk1, 2019-07-02:
Motor Mixer
Motor	Throttle	Roll	    Pitch	     Yaw	
1	1 -0.42100000000000004 1 0.8999999999999999

2	1 -0.5800000000000001 -1.1 0.8889999999999998

3	1 0.41999999999999993 1 -0.8999999999999999

4	1 0.5800000000000001 -1.1 -0.8899999999999999

5	1 	-1 	0 	-1

6	1 	1 	0 	1

rrk1, 2019-07-02:
Well, most values look relatively benign, like round-off from int/float conversion or so, but math with motor 1/roll looks somewhat suspicious when 0.421 is substituted to 0.42.

DzikuVx, 2019-08-30:
This is float->int->float rounding effect. You can safely use those values without any problems

closed 2019-08-30 by DzikuVx

rrk1, 2019-08-30:
I have to argue. Until mixer values are stored internally as 8 bit, there is way too much error. 0.421 vs 0.42 is +0.24%, too much for signed int16 rounding, somewhere may be a bit loss or other flawed math.