iNavFlight/inav-configurator #802: Nav modes do not activate

fremebondo, 2019-07-03:
I have the same identical problem reported in  #597  (Closed without a documented solution).

I'm trying to activate nav modes for my fw airplane on the ground before going up in the air.
I get Nav Altitude Hold BLUE on Modes page, but no other Nav mode became blue (Cruise, PosHold, RTH).

I have a good GPS fix (17 sats, 1.3m HDOP), I can arm the craft and fly, I also tried loading a WP mission setting a couple of WPs around my position. No luck with activating Nav modes.
 
Anyone faced a similar scenario?

DzikuVx, 2019-07-03:
NAV modes can be activated only after arming hence you can not activate on the ground. This is not a bug, it's by design

closed 2019-07-03 by DzikuVx

fremebondo, 2019-07-03:
I activaded AFTER arming

fremebondo, 2019-07-03:
I arm the craft, then switch on NavPosHold and it does not become "blue"