iNavFlight/inav-configurator #803: Nav modes do not activate AFTER ARMING

fremebondo, 2019-07-03:
Again, opening as support related to #802, closed without letting me reply:

When I try to arm PosHold with a GOOD 3d gps fix, and the craft ARMED, PosHold does not become "BLUE", nor Cruise or other NavModes. Only AltHold become "BLUE".

If is not a bug, please wait and close the issue after a solution is given and **confirmed by the reporter**, so it can be usefull for other people facing the same problem.

digitalentity, 2019-07-03:
NAV modes won't active until correct heading is acquired. Move your plane forward at speed over 5-6m/s

fremebondo, 2019-07-03:
Ok, that's could be a good point. I will try it in a few hour and let you know.

fremebondo, 2019-07-03:
By the way, i suggest that those "hidden" conditions could be documented or, better, notified within Configurator in a manner similar to "pre-arm checks"

fremebondo, 2019-07-03:
And in https://github.com/iNavFlight/inav/wiki/Navigation-modes I read:

> Hints for safe operation:
> Activate without props installed to check for reasonable operation.

which i assume I have the craft still on the ground, as it cannot move without a prop!

Of course I can try carrying while walking, but IMHO is not clear in the wiki,

fremebondo, 2019-07-03:
Hi @digitalentity, your suggestion is right, nav modes works while in flight. 
NOW the issue can be closed, but I recommend to document this behaviour in the wiki!

closed 2019-07-16 by DzikuVx