iNavFlight/inav-configurator #808: Mode selection, lowest is CH5

Jetrell, 2019-07-14:
Presently the lowest channel that can be allocated to a mode selection is channel 5.
This is fair enough in multicopters and conventional fixed wings.  
But it ends up being a wasted channel, when used with a flying wing (no rudder).

If the mixer page has channel 4 (normally rudder) not selected. And the Yaw PID values are set to zero. 
Then channel 4 could be used for a mode selection. 
Considering that most micro RX's only have 12 channels, of which up to 4 are control and 1 RSSI/LQ and 1 is arming. This narrows down the use of many of the modes that the user may want to select. So it would be good, not to waste channel 4 in a flying wing air frame.

I also think it would be an idea to have the YAW mixer and the lowest mode channel number tied together. 
So that when a mixer is selected for Yaw. Then channel 5 remains the lowest channel.
Channel 4 would only appear under the modes tab, if NO Yaw mix is selected.

KeenFPVer, 2019-07-14:
You need rudder channel to use OSD and change settings correctly.

Jetrell, 2019-07-14:
@KeenFPVer Do you mean the CMS menu's?  That's true. 
But its only an option if the person chooses to select channel 4 on a flying wing, for another purpose.
They would also have to forgo using the CMS.  
It would not affect Quads or Normal fixed wings planes. And even then, would only affect a flying wing, IF the user decided to used channel 4 for another purpose.

DzikuVx, 2019-08-30:
4 first channels have reserved functionality and as such can not be used to activate modes

closed 2019-08-30 by DzikuVx