iNavFlight/inav-configurator #809: Compass calibration IST8310 does not work with inav 2.2 RC2

Ralfde, 2019-07-14:
The calibration does not show the 30 seconds any more.
Calibration does not work.
After it stopped the values are still 0 after save and reboot.
The compass seem to work. Just the graph is not shown correctly, because of the offset.

[image]

digitalentity, 2019-07-14:
The same issue also reported by @Ralfde via email.

devalexqt, 2019-07-16:
I have the same issue too.

VydioFeed, 2019-08-16:
Hi, Long time listener, first time caller. I just finished a completely new build and seem to be experiencing the same, or almost same issue. Using 2.2 configurator and firmware, the mag is green lit in the upper hardware list at the top of the page but will not calibrate. I'm using the 722 SE FC and a new Holybro pix32 neo m8n with the IST8310 on board. GPS is working flawlessly, 16 Sats in the house. Originally I thought it might be the wiring but after firing everything up, the config. found and correctly labeled the mag as the IST when put to "auto" mode. I also noticed that when I first went to the sensors tab the mag was showing movement as I moved the quad around. After going back and trying to re calibrate several times (with the same results as Ralfde) with no success (all "0"s), I then went back to the senor graphs only to find the mag no longer showed signs of life... just flat lines.  The hardware icon is still green and the IST still shows in the hardware list. This is a real bummer as my $1500 build is now a paper weight until I can sort this out. Hoping maybe someone has some input? This is my 10th full build, so I'm confident in that department and I nailed everything else on the first try. hoping is software and I don't have to tear this thing down. : (

Ralfde, 2019-08-16:
You can store the compass calibration manually.
After doing the compass calibration go to cli and use
save
to store the compass calibration values.

After saving you can check if they are correct.

VydioFeed, 2019-08-21:
Hey Ralfde, thanks for the quick response. I did find some info on stick calibration, but not much. I actually decided to just go back to the hardware configuration and realized I made a pretty noob mistake. In a rush to complete the build i simply mounted the gps to the mast to get it up in the air without any concern for the directional arrow. Apparently the FC does not like it at all if the antenna is not facing the correct direction. I figured the sensors would still get picked up anyway but this seems to not be the case for the mag. Once corrected everything came to life.  
        I do however have one more issue that I have been pouring over for a few nights now without any solution.  Using the R9M system ( fully updated at both ends) with the 722-SE board I can't seem to get all channels to respond in the receiver or mode tabs. Channels 6, 7, and 8 are unresponsive but 5 and 9 work fine ( sticks are operational as well, RSSI on 16).  I've searched for anything related to aux channel adjustments, or enabling aux channels ( like in Betaflight ) but have come across very little. I updated the Matek to 2.2.1 (no change from 2.2.0 ) and the Qx7 is also updated. All switches are working correctly on the TX side. Is there a command to enable or disable channels in the cli? I feel like I'm missing something obvious, but it is eluding me. Sorry to post this question here but I haven't found anyone with a similar issue. Thanks in advance.

VydioFeed, 2019-08-21:
Edit... I started a new issue concerning aux channels.

closed 2020-06-11 by DzikuVx