iNavFlight/inav-configurator #816: Configurator 2.2.0 & 2.2.1 SPRACINGF3 flashing issue

Philippe-France, 2019-07-28:
hello
can't flash SPRACINGF3 with configurator 2.2.0 & 2.2.1

configuration:
windows 10
configurator 2.2.0 & 2.2.1
SPRACINGF3 with iNav 2.2.1
Baud rate set to 256000
other options disabled

procedure:
USB cable unpluged, start iNav configurator
plug USB cable : COM4 appears in COM selector
go to flash sheet
unselect all options except set manual baud rate to 256000
select local FW iNav 2.2.1 for SPRACINGF3 "inav_2.2.1_SPRACINGF3.hex"
click flash firmware: after #10s get the message "Failed to flash COM4"
-- same error when selecting board & FW in dropdown field --

Works fine with BetaFlight configurator 10.5.1 loading local FW and Baud rate set to 256000 (no other option enabled). New flashed FW type & version checked on iNav CLI: ok.

Note: procedure repeated a second time on same board: BetaFlight configurator is able to reflash the board where iNav 2.2.1 is present, while iNav Configurator can't flash.

regards

AkinakaTone, 2019-08-11:
Information of my experience.

I encountered the same problem for my SPRACING3 today, with hex file previously save on PC.
Using baud rate 115200 (my usage habit). I attempted to use configurator  2.1.0, with similar failure.  Then used ver 2.1.4, downloaded from home page and worked. Reason unknown.

And then used 2.2.0 to configure my hex copter with success except mag calibration, as pointed out in this issue list. But as explained inside the issue, used 2.1.4 again to calibrate as before.

DzikuVx, 2019-08-30:
try recovery procedure with `boot` pins shorted. I flashed SPRF3 a few weeks ago without problems

closed 2019-08-30 by DzikuVx

AkinakaTone, 2019-09-01:
Hi Pawl-San

 

Thank you very much for your information.

I will try at next time.

 

I am still not familiar with the use of boot pin and still learning STM32.

 

However, just used one time desperately, a few days ago on STM32H747 DISCO, just bought recently.

Most of the IDEs (Atollic TrueStdio, STM32CubeIDE) still do not support STM32H747 dual core (480MHz) MPU.

 

On this board, I accidentally (but with some intention) loaded STM32H743 program and hung/garbled, encountered “No target found!” error.

Therefore, used solder bridge on R192 to 3.3V, successfully connected, erased flash, removed solder bridge and then program load succeeded.

 

Sorry for annoying extra description. 

Again, thank you very much.

 

Akio Nakamura