iNavFlight/inav-configurator #825: Some aux channels (6,7,8) not responding in mode or receiver tab, Matek 722-SE .

VydioFeed, 2019-08-21:
Using the R9M system ( fully updated at both ends) with the 722-SE board I can't seem to get all channels to respond in the receiver or mode tabs. Channels 6, 7, and 8 are unresponsive but 5 and 9 work fine ( sticks are operational as well, RSSI on 16). I've searched for anything related to aux channel adjustments, or enabling aux channels ( like in Betaflight ) but have come across very little. I updated the Matek to 2.2.1 (no change from 2.2.0 ) and the Qx7 is also updated. All switches are working correctly on the TX side. Is there a command to enable or disable channels in the cli? I feel like I'm missing something obvious, but it is eluding me.  Thanks in advance.

Mateyhv, 2019-08-26:
I might be wrong but doesn't seem to be a problem in Inav but rather an issue on Frsky side. Have you tested the receiver in another setup to exclude possible faulty equipment? I have the same setup using the R9 Slim+ receiver, however I have only tested F.port, Flex firmware D16 (1 to 8 channels) and had no problem with any auxiliary channel.

DzikuVx, 2019-08-27:
This is for 99,99% caused by wrong configuration of the radio -> it is not configured to output those channels to the receiver

closed 2019-08-28 by DzikuVx

xoga74, 2021-12-03:
I had the same problem en by lowering the output from 100% to 80% on my radio, channels started to respond again.  With setting 100% it seemed inputs where out of range of correct inputs for flightcontroller.  In betaflight you can change these values in failsafe tab, i didn't find this option in Inav.

VydioFeed, 2021-12-04:
Hi, ya in the end it was something very simple and i felt stupid. This was
a year ago or so, but I believe I had
bound my receiver with some of the upper switches (four at top face of Qx7)
in the neutral range. So when
i went into the receiver tab of INAV those particular switches didn't work
but others did. It was by sheer accident
that I had bumped one of the switches into a different position (up or
down) that allowed that channel to start responding
again after another re-bind session. I don't know if this is an Inav issue
or a radio issue....
Scott Jennings
Vydio Feed Inc.