iNavFlight/inav-configurator #826: accelerometer calibration sequence resets after upside down step

humanoid2050, 2019-08-30:
I'm running inav and inav configurator 2.2.1. The board is Matek F722Mini, but I don't think that's relevant. When I do the Accelerometer Configuration, I can position my quadcopter and the orientation will be automatically detected and logged as expected, right up to the point where I do the upside down orientation. The orientation will be detected, and all previously logged orientations will be simultaneously forgotten. It's easy to just do the inverted orientation first if you know, but its a bit confusing until you find the pattern.
Issue #586 might be related, hard to tell from the description provided.

ds7060, 2019-10-31:
I have the same issue with the 722 Wing, MAtek just told me to repeat step 1 at the end, it works.

DzikuVx, 2019-11-04:
Board has to be stationary, not moving, not shaking, stable while calibrating. When one of the step fails, you just have to repeat

closed 2019-11-04 by DzikuVx