iNavFlight/inav-configurator #831: Inav APT repository

ramoonas, 2019-09-20:
Hi,

I have couple of not too overloaded linux servers available and i could host APT repo for deb stable and nightly packages. In fact, i already have one running on linux container here http://apt.cloudpilot.lt/Packages because i am lazy to install inav, mwp and betaflight to couple of my laptops plus rapsberry pi manually (you can use it if you wish by adding "deb [trusted=yes] http://apt.coudpilot.lt/ /" to sources.list.

However, if people think it would be useful i suppose i could host inav APT repo on a faster server in Netherlands or US and differently to this one I could set it up with GPG signing etc.

Do you think it would be useful?

Furthermore, if you have any other need for servers (i.e. automatic nightly build for inav-configurator or inav firmware or even hosting jenkins) feel free to comment. I can arrange that

DzikuVx, 2019-09-23:
Hi, thanks for contacting us. Could you contact me on pspychalski@gmail.com or on Telegram @dzikuvx directly?

ramoonas, 2019-09-23:
sent you an email

closed 2021-06-26 by DzikuVx