iNavFlight/inav-configurator #840: No Matek F411 WSE target in Inav configurator 2.2.1

benanerson, 2019-10-06:
Hello,

While trying to set up my Matek F411 WSE flight controller, I noticed that their is no target for the Matek F411 WSE flight controller. On Matek's websie they specify the target as "MATEK411SE" however this target does not appear in the Inca configurator. 

Thank you! :)

[image]

DzikuVx, 2019-10-06:
This target will be released as INAV 2.3

closed 2019-10-06 by DzikuVx

benanerson, 2019-10-06:
Hi DzikuVx,

Thanks for getting back to me. Is their an estimated release date for INAV 2.3 is? 

Thank you.

unluckyfrog, 2019-10-25:
I've loaded 2.3 this morning and there is no Matek f411se target. Is there another target i can use?

Grumpyrse, 2019-11-22:
Any update on this?

DzikuVx, 2019-11-22:
@Grumpyrse INAV Configurator 2.3 and latest RC od INAV 2.3