iNavFlight/inav-configurator #852: Matek F722 and F405 permanently stuck in STM32 FS Mode after flashing 2.3 RC3

FelixvdDonk, 2019-11-03:
I flashed both the F722 and F405 to the newest release candidate with 'Full chip erase' enabled.
Now both of these flight controllers present themselves as 'STM32 virtual com port in FS Mode'. 

As seen on Matek's website: http://www.mateksys.com/?p=5159  
FS mode is not suitable.

Solutions tried: 

- Flashing to older INAV 2.2.1 with and without 'full chip erase'
- Flashing to older INAV 2.2.1 with older INAV configurator
- Uninstalling all STM32 drivers, rebooting and following Matek's website instructions
- Manually changing the driver from STM32 virtual com port in FS Mode to STM32 virtual com port
in device manager.

It seems like this is not a driver issue and the boards themselves are behaving differently.
I cannot find a good explanation as to what FS Mode entails. I really hope these boards are not bricked right now

FelixvdDonk, 2019-11-03:
I see I haven't fully described the issue. I am able to flash the Matek F722 and F405 in DFU mode however they will not respond when trying to 'connect' with INAV configurator. The COM port is visible but there is no response when trying to connect. The devices now show up as 'STM32 virtual com port in FS Mode' in Zadig, which they previously did not

FelixvdDonk, 2019-11-03:
Problem solved. I was confused with the target I needed to flash. Given the Matek F722 SE is a different product from the F722 wing, I assumed I needed to flash F722.

closed 2019-11-03 by FelixvdDonk

shrinibas7, 2024-09-24:
i was able to go DFU mode in inav,but after flashing the firmware the COM port not showing can anyone help.......

shrinibas7, 2024-09-24:
> i was able to go DFU mode in inav,but after flashing the firmware the COM port not showing can anyone help.......

i am using stm32 blackpill and type C cable direct to board

sensei-hacker, 2024-09-24:
> i was able to go DFU mode in inav,but after flashing the firmware the COM port not showing can anyone help.......

Are you running an ancient version of INAV?
On a Matek F405?
Unclear on FS mode?

If the issue you are experiencing has absolutely nothing to do with this issue, please create your own, with the requested information from the template about what you are using ans ehat your problem is. Thank you.