iNavFlight/inav-configurator #854: 2.3.0 RC3 Mission Control Problems

breadoven, 2019-11-05:
Noticed 2 issues with Mission Control in Configurator 2.3.0 RC3 (Win 64):

1. The File Open button does nothing ... no file open dialogue box appears.

2. WP's don't seem to want to save in EEPROM, least not permanently. If I enter some new WPs then save to FC board (Mission Valid red box turns green) then Save to eeprom all appears OK. Then go into CLi and type WP and the new WPs appear as expected. However, if I go back into Mission Control after board has rebooted you can't load the WPs from eeprom and when you go into CLi and type WP the WPs have reset and disappeared. There's a message saying WP0 is invalid,

This was working under 2.2.1 but not now.

INAV version is OMNIBUSF4PRO_LEDSTRIPM5 2.3.0 RC1 (custom build to fix Baro problem).

breadoven, 2019-11-05:
Actually after messing some more it seems that even though it's showing a message "EEPROM saved" in fact it hasn't saved it, there is no confirmatory beep. It's just saving to the FC board but not eeprom.

breadoven, 2019-11-07:
After checking loading waypoints using stick commands it seems that waypoints are being saved to eeprom and can be loaded using stick commands. It's just Configurator that won't load using the eeprom load button. After you load using stick commands and then go into Mission Control the waypoints appear to load automatically in the waypoints available text (and the status turns green) but nothing is shown on the map itself. So sort of works but a bit buggy if you try and load WPs from the board.

Dronitoni, 2019-11-15:
I see the same, that the **"load file"** button is not working.
I'm using the Configurator 2.3.0-RC3.

Dronitoni, 2019-11-20:
I have again some trouble with the load file button. 
Now I'm using the newest configurator 2.3.0, but the load file button doesn't work.

breadoven, 2019-11-20:
Yes, 2.3.0 still doesn't work for me. No file open dialogue displaying, no Eeprom load. Seems a bit odd because the code for file open doesn't seem to have changed and uses the same method as the file save button (which does work with me). My programming skills are limited to VB so don't understand JS well enough to figure out why this would be. Only work around is to save and load mission WPs from CLi by copy and paste.

DzikuVx, 2019-11-20:
Load button will be fixed in #859

DzikuVx, 2019-11-20:
1. Load from file will be fixed in 2.3.1
1. Save To EEPROM work only **after** _Save to FC_ was used. It simply persists what _Save to FC_ stored
1. After reboot it is required to use _Load from EEPROM_ to load a mission. List stored in EEPROM is not the list that will be used for mission

closed 2019-11-20 by DzikuVx

breadoven, 2019-11-21:
Just to confirm that version 2.3.1. had fixed all the above problems including eeprom load and waypoints not displaying. Now all working as 2.2.1. Thanks for that.