iNavFlight/inav-configurator #855: RTH flies to position but drops several feet before landing speed kicks in!

rholt51, 2019-11-09:
i'm pretty sure this is a setting issue i have with my 7" TBS Source One using a Matek 405-CTR FC and Taranis x9D plus.
i have tuned my receiver for throttle down to 75% because i'm a beginner and the 100% really overdoes my proficiency.

everything works fine with iNav including Position Hold, Alt Hold and flight modes.
but when initiating RTH, the quad:
1) correctly flies to altitude
2) turns towards landing point
3) flies back to landing point
4) turns to the position of takeoff.
5!!! Drops about 15 feet before the motors speed up and then brings the quad to a correct landing.

I have used RTH before, but the drop is new. i've tried to look up this problem but cannot find this issue taken up with anyone else. 
your help is appreciated...
Russ from Ocala Florida, USA

DzikuVx, 2019-11-22:
probably hover_throttle issue. You have to check in blackbox log what is happening and if so, this is for the firmware part, not configurator project.

closed 2019-11-22 by DzikuVx

rholt51, 2019-11-24:
thanks, I will do that and repost what i find.
russ from Ocala, Fl, USA

rholt51, 2019-11-25:
I have since up’d the Throttle  from 50% to 75% on my Taranis and it seems to have improved the amount of drop to a large extent. 

My hope is to get better “acquainted” with my 7” build (F722-SE, R-xsr, RaceStar 4-in1, RaceStar BR2508’s, TBS Unify Pro, on a TBS Source One with the 7” arm upgrade. Updated the R-XSR to utilize FPort.

RunCam Split mini. 

A very newcomer to FPV, although I’ve had my FS Dominator HD’s for over 2 years, now!

I’ve taken a lot of queues from your many (and valuable) videos over the past couple of years. 

Please keep up the great work!

Russ from Ocala, Fl, USA

 

About to go fly tomorrow with the Source One and my “newly” rebuilt Lumenier QAV250 <-- yes, an oldie that has been on the shelf for some time now. It was my 1st build but it has seen many rebuilds and upgrades as I’ve gone along this path.