iNavFlight/inav-configurator #856: iNav 2.3RC3	GPS shows blank sheet instead of Map

BenWThomas, 2019-11-11:
## Current Behavior
No map is displayed at all; current location marked on an empty frame

## Steps to Reproduce

1.   MacOS 10.14.6, iNav Configurator 2.3RC3, FW iNav 2.3
2.  Matek F722SD FCU, RadioLink UBLOX 8 + I2C compass
3.  Configure, calibrate (CW 90° Flip Mag req), connect USB 
4.  Wait for sat lock (9-14 at bench top)
5.  Go to GPS tab

## Expected behavior
After GPS has determine location, map of area with flagged location should appear

## Actual behavior
Everything looks fine, except there is no map: Quadcopter location icon appears, centered
Clicking +/- to zoom shows a little displacement of Quad icon, as expected
   Seem to be able to zoom + several more increments than expected from usual default display
   At very high zoom, quad flag drift can take it completely out of the [blank] viewing area
      Previously working display at max map zoom drifts only a few meters, over time, at most
         (drift << 10% from center of displayed map over time, in previous releases)

Enclosure:   
[image]

## Suggested solution(s)
I think my problems with GPS map display began with iNav 2.1; I don’t think it has anything to do with my hardware or settings.  Everything was fine before (as I recall) 2.1.  Test setup isn’t calibrated, but also displays no map after GPS lock.  9 sats on my desktop, 3D lock, no map in 2.3.x; map but no zoom in prior recent releases.

I tried with a new account (so a new home), which weirdly tried to redirect to my own home but was blocked by permissions.
I also updated to Configurator 2.3RC3 on a system that had only ever had 2.2.1 installed previously; same behavior.

## Comment
I’ve seen a few others chat about problems with GPS map in these release, too.

## history
I think my problems with GPS map display began with iNav 2.1; I don’t think it has anything to do with my hardware or settings.  Everything was fine before (as I recall) 2.1.  Test setup isn’t calibrated, but also displays no map after GPS lock.  9 sats on my desktop, 3D lock, no map in 2.3.x; map but no zoom in prior recent releases.

DUMP    https://pastebin.com/WU3yNP7T

---

- FC Board name and vendor:  Matek F722SD
- INAV version string:   # version
# INAV/MATEKF722 2.3.0 Oct 26 2019 / 19:39:51 (630430526)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

leeph, 2019-11-18:
I have this issue as well. It also transpires that the 2.3 configurator release candidates both do not successfully pull the list of boards & firmware targets from the internet - the dropdowns stick on 'Loading...' and never populate.

I have had to download 2.3 RC hex targets manually and flash.

Running 2.2.1 on the same machine yields no problem retrieving the target options, so it's not a connectivity issue.

DzikuVx, 2019-11-18:
Please check Map configuration and that there are no firewall or other limitations from macOS side to OpenStreetMaps or Bing maps. On my side seems to be working perfectly fine
[image]

[image]

[image]
[image]

BenWThomas, 2019-11-25:
Hi, Pawel,

Thanks very much for pointing the way.  

Many releases ago, I’d configured Bing maps (which worked fine, once I registered) then switched back to OpenStreetMap as my preference.  OpenStreetMap continued to work fine, until I upgraded to iNav 2.0.x, then, suddenly, it wouldn’t hold zoom changes.

Following your lead, I checked my map preferences, and everything looked fine.
I changed it to Bing, found zooming worked fine there, then switched back to OpenStreetMap, and.. lo and behold! Zoom worked fine again.

=>	While there was no net change visible in the Map Preferences settings, it appears that re-writing my preference cleaned up something.

Thanks again!

Ben

closed 2019-11-25 by DzikuVx

rholt51, 2020-02-09:
Hmmm...
20200209.... WIndows 10, INav v2.4.0
Matek - fs722 Beitian bn-880
8+ satellites.
no MAP on GPS just the spotter icon that moves around when i zoom in .

BenWThomas, 2020-04-21:
Hi, Russ.

In iNav, it's on the little Gear Thing at uppermost right: try changing your Map Provider from OpenStreet to Bing; save, restart iNav.  Change back to OpenStreet and save/restart iNav again.

(Not sure if I recall that the restarting of iNav is necessary, or not, but in my case, it was just the settings for my Map Provider that were pooched.  Perhaps because a few few releases back, I had briefly changed to Bing and got the API key from Microsoft and all that... didn't like it, changed back to OpenStreet.  OpenStreet worked fine at the time, but some iNav update or another seems not to have liked what I'd done when I switched back and forth.)

Better?

rholt51, 2020-04-21:
Ben!!!!

I’ve have literally spent HOURS on this thing – even to the point of trying to set up that API route!

Not to mention ALL the sites that I searched & scoured to get a clue.

Finally, your email came and I feel like all is well with the world (outside of the virus, that is).

I hope you are well, and thank you SOOOOoooo much for your email!

Btw – the map came right up after I set it to Open Street.

 

Russ from Ocala, Florida.

 

This email has been checked for viruses by AVG.
https://www.avg.com

rholt51, 2020-04-21:
> Russ, this was a huge headache for me, too. I spent many hours, over several months, before I broke down and reported the bug. I began having the problem with 2.1, and didn’t get it fixed until 2.4...
>
> If it was addressed as a bug, perhaps it cannot occur any longer with a new release, but my frantic, then resigned, testing suggests that the data we reset when we did what we did to restore OpenStreet maps lives somewhere in support files or my home directory; I never was able to get iNav to show maps on that boot volume until I got Pawel’s (github ID = DzikuVx) response to the bug I filed.
>
> Pawel couldn’t reproduce my problem, but he focussed my attention on Map configuration.
> That triggered ancient troubleshooting memory: "If you cannot see or locate the settings to pitch them, then re-write them.”
>
> I figure it was probably some path that broke, perhaps by me moving a directory.
>
> Please tell your friends!
>
> Regards
>
> Ben Thomas
> Austin

HAH! Often, that is the way I fix a bug, also!!!

In any case, so many hours wasted between the two of us – and I’m SURE there are many others. As I go out to the sites I posted my problem to, I’ll update the CORRECT (and so easy) way to fix their dilemma.

Thanks once more.

Btw, I watch everything that Pawel puts out there. I’m very interested in obstical avoidance and I’m hoping that Pawel will keep moving forward with this using the MatekSys Optical Flow Lidar Sensor. Mine should be delivered this week.

 

Russ from Ocala, Florida, USA