iNavFlight/inav-configurator #863: Configurator 2.3.1 Font manager selection issue

Jetrell, 2019-11-21:
The font manager in 2.3.1 works intermittently when selecting a new font. 
I tested it multiple time with alternate Configurators and firmware's with my Window 7 32bit PC that I take to the field.
The FC and Configurator have to be restarted to get a selection. And it doesn't seem possible to `open font file` at all.

[image]

As a side note. This RED square below is now the toolbar icon for iNav.

[image]

DzikuVx, 2019-11-21:
OK, another issue with changes in nw.js and the fact that API over there is now changing...

DzikuVx, 2019-11-23:
new configurator will be released very soon so closing

closed 2019-11-23 by DzikuVx

Jetrell, 2019-11-24:
> 
> 
> new configurator will be released very soon so closing

Thanks Pawel.  I tested 2.3.2 and it works great :+1: