iNavFlight/inav-configurator #870: Servo number mismatch in configurator 2.3.1

Arvycka, 2019-11-22:
Servo numbers in "Output mapping" section do not match servo numbers in "Servo mixer" section when a mixer preset is loaded.
[image]

Why do we even need these servo numbers? Couldn't we just use Ouptus? It complicated the mixer unnecessarily.

DzikuVx, 2019-11-22:
They are not supposed to match. What you define in table is just an arbitrary number than later is converted to servo output number and actual output.
On top of that, some internal functions relay on servo index. 
If you want numbers to match, index them from 0.

closed 2019-11-22 by DzikuVx

Arvycka, 2019-11-22:
But now that's even more confusing than it was before - now we have 3
numbers for a servo, an output, a number below it and a number in the
mixer. CLI uses mixer numbering. Output servo configuration also uses mixer
numbering, but outputs are paired with different non matching numbers? How
to track which servo is doing what?
The presets should be updated too and not to assign a random servo number
starting with 2
Also, at least in 2.2.1, servo autotrim ONLY works with default mapping
from a preset. You can number the servos whatever you like and control and
stabilization works fine, but servo autotrim only works correctly with
default mapping. Has this been fixed in 2.3?