iNavFlight/inav-configurator #884: [Linux] Application-Window not resizable

Neuroplant, 2019-12-23:
Application-Window is too big for my Notebook.Screen (1368x768) and not resizeable.
So I can not use it, buttons on bottom not reachable
(OS:Ubuntu 18.04.3 LTS Desktop)
Hope you can fix this
Neuroplant

stopdick, 2019-12-30:
Same thing here! It appears that the minimum size for the inav configurator window is "1024x800". Running it on my notebook which has a 13.3in (1366x768) screen hides the bottom of the window. Its pretty important because i can not see and use the "Save" button. I also don't get the option to maximize the window.

A work around is to move the window up by pressing the "Alt+F7" keys and move the mouse (or hold "Alt"+Click the window and then move it, depending the system you are using).

[image]

rsilk1949, 2020-02-24:
Same problem on WIndows 10 laptop with screen 1366x768

rsilk1949, 2020-02-24:
> Same problem on WIndows 10 laptop with screen 1366x768.
Can be "fixed" by maximising window each time I open the app.

Neuroplant, 2020-02-24:
@rsilk1949 this is not the same problem as the window IS resizeable under Win10.
It is just a little inconvinient

DzikuVx, 2020-03-03:
#942 min windows size was lowered to 800x600

closed 2020-03-03 by DzikuVx