iNavFlight/inav-configurator #887: 2.3.3 sets throttle values to default

Jetrell, 2019-12-31:
The 2.3.3 configurator sets all throttle output values to <default, the first time it connects to a model with 2.3 installed.  I have noticed this with every model I have connected to it.

I understand that this is a work in progress with the `min throttle/idle` addition. 
But its a time trap to chase down the issue.  A warning on the download page would be nice :-)

Thanks
[image]
[image]

DzikuVx, 2019-12-31:
That was fixed in the release. Or at least I was sure I fixed that. Are you
using latest downloadable 2.3.3?

DzikuVx, 2019-12-31:
just checked, I fixed it in https://github.com/iNavFlight/inav-configurator/commit/947ec987f92a7c694dd0eef9c6b89bafc9976180

Jetrell, 2019-12-31:
Thanks for the quick reply Pawel.
It sure looks to be fixed :+1: 
I downloaded the 2.3.3 configurator yesterday and used it the first time today. This issue happened on several flight controllers.
But after going back over it just now and re-flashing one of the boards a few times with 2.3.2 and 2.3.3...
I can't replicate it again :-/

Sorry for the false alarm. I'll close this now.
Thanks again

closed 2019-12-31 by Jetrell

DzikuVx, 2019-12-31:
It was happening when you saved anything on „Configuration” page. Check that again please if you can Paweł Spychalski Od: JetrellWysłano: wtorek, 31 grudnia 2019 11:47Do: iNavFlight/inav-configuratorDW: Paweł Spychalski; CommentTemat: Re: [iNavFlight/inav-configurator] 2.3.3 sets throttle values to default (#887) Thanks for the quick reply Pawel.It sure looks to be fixed 👍I downloaded the 2.3.3 configurator yesterday and used it the first time today. This issue happened on several flight controllers.But after going back over it just now and re-flashing one of the boards a few times with 2.3.2 and 2.3.3...I can't replicate it again :-/Sorry for the false alarm. I'll close this now.Thanks again

Jetrell, 2019-12-31:
@DzikuVx  I have been over it again and tried loading the 2.3 firmware with the 2.3.2 configurator. I also went back and loaded a couple of different `diff all `files and did a merge check, to make sure nothing was left out. All looked clean.

With respect to this board OmnibusF4. I only flew that quad two days ago and every thing worked fine. But I did change some PID's and filters settings and downloaded the logs after that flight.
Then today I plug it into 2.3.3 and changed an OSD element position. 
Then later when I went to fire it up, the motors wouldn't start. Then I remembered the throttle values that I had issues with yesterday, on the Matek F7. And that was the problem.. they too where at min levels. 
But can even work out how the two are related. Two are on fixed wings and the other a quad.

So now I can't replicate it. Don't know if that's good or bad. Lol
If it happens again, I will report back to you here.
Thanks