iNavFlight/inav-configurator #907: Load firmware [local] not working

Aaronprat, 2020-01-25:
hello,

When trying to flash a new firmware through iNav configurator, loading the firmware locally is not working on version 2.3.3.

The only time it's working for me is version 1.9.1.
I'm working with OSX's latest version, tried from multiple mac computers.

That's probably not enough information. In this case, I'm not sure what else I can share, so please lmk if you guys need any other info.

Thanks,
Aaron

machadofelipe, 2020-02-01:
Same problem here with the 2.4 and 2.3 versions, but in my case it seems that the configurator does not understand the bootloader mode. Even if I go into manually and set for no reboot.
My workaround is to download the asset manually and use betaflight config. to flash inav :)

DzikuVx, 2020-03-03:
I belive this was fixed before 2.4 final

closed 2020-03-03 by DzikuVx

machadofelipe, 2020-03-03:
Tested the 2.4.1 the other day and had the same result. How can I debug this?

sensei-hacker, 2025-07-05:
> I tried to flash my own compiled inav firmware and both in inav configurator 7.1.1 and 8.0.1 was impossible. Pressing Flash did nothing. I had to flash it with betaflight configurator, which is a crazy irony because I want to go away from betaflight!!
> 
> Please reopen the issue!!!

First, use the DFU button on the board to set the board to DFU mode. Then on the flashing page set it to not reboot.

If that works, you can also test flashing without using the button by having the option to not reboot on, and see if it works to click the flash button, wait about two seconds, then click it again. Configurator may not be giving it quite enough time to reboot to DFU mode.

Also, as Mr D said, if there is an issue, it's probably a new issue. :) Probably different from the thing that was fixed five years ago.

cptX, 2025-07-05:
I just found the issue. I'm in linux and my make command created an .elf file in the bin folder, and I tried to convert this file in hex with another tool. This hex file could not be flashed with inav, but was successfully flashed and worked by using betaflight. I just discovered though that the make command created indeed a .hex file elsewhere and this was able to be flashed from inav. Probably another file structure or some labeling in the file that didn't allowed it to flash when I did the conversion from .elf. Anyway please close the issue and sorry for the inconvenience