iNavFlight/inav-configurator #913: Wrong Output Mapping in mixer tab

jreise-d, 2020-01-31:
iNav configurator 2.4.0 with iNav 2.4.0 RC2 on Matek F411 Wing

After loading an airplane Mixer preset in the mixer tab it shows a wrong output mapping. See the attached screenshot, it shows servo3 and servo4 twice. 
[image]

jj-uk, 2020-02-02:
Can you take another picture showing more of the config screen? Further down there should be the servo allocations. Servo 3 and 4 are used for flaps and stabilised roll at the same time.

jreise-d, 2020-02-02:
> Servo 3 and 4 are used for flaps and stabilised roll at the same time.

That's right. The servo allocation in the servo mixer is correct, smix in cli is correct, iNav also works correctly. But still servo3 should only be connected to output S4 and servo4 to S3 and so on, so the Configurator display in the Output Mapping is wrong. This is wrong in all airplane presets in the Mixer Tab, e.g. here in the airplane V-tail:
[image]

jreise-d, 2020-02-25:
This error in the output mapping section of the Mixer Tab also exists in the new iNav 2.4.0 and configurator 2.4.1. Tested on Matek F411-Wing and F405-Wing

jj-uk, 2020-02-25:
Yes, I ran into this on the Matek F722-wing the other day. I reset the servo numbers to begin at 1 instead of 0 and that worked for me, but it does seem to be a bug.

mimo55, 2020-04-30:
I have the same problem with a V-Tail fixwing.
There are too many servo outputs in the configurator.

jreise-d, 2020-05-19:
Tested with inav 2.5.0 RC1 and configurator 2.5.0 RC2, the error remains. Occurs when flaps or multiple mixers act on the same servo output.

jreise-d, 2020-06-09:
Tested with new configurator 2.5.0 and inav 2.5.0, same error
[image]

It's a pity that none of the developers look here. The bug exists since January and version 2.4.0 and is easily reproducible. A fix is surely only a small thing

mrcottonmouth, 2020-07-15:
Any ideas how I should set up mu alba bird, the mix looks messed up in the configurator. I should configure this manually in the cli, but not sure where to start. Any example smixes to review?

wx4cb, 2020-08-31:
bump? anyone fixed this yet? or have a work around? my pan/tilt doesn't work until i can figure out where to plug the darn thing in

DzikuVx, 2020-10-26:
will be fixed in 2.6

closed 2020-10-26 by DzikuVx