iNavFlight/inav-configurator #914: [REQUEST] Add AR Wing preset to configurator

b14ckyy, 2020-02-02:
This is my manual tune for the AR wing with stock servos and optimized for speeds between 50 and 70 kph for a medium powered cruise setup or stock setup. May be a nice preset for the AR wing. 

set fw_p_pitch = 4
set fw_i_pitch = 15
set fw_ff_pitch = 50
set fw_p_roll = 5
set fw_i_roll = 10
set fw_ff_roll = 40

set roll_rate = 28
set pitch_rate = 12

These PIDs are for the following Servo Mix as the default 50% mix is too low on that plane: 

# servo mix
smix 0 3 0 80 0 -1
smix 1 3 1 60 0 -1
smix 2 4 0 -80 0 -1
smix 3 4 1 60 0 -1

DzikuVx, 2020-02-06:
see #919

closed 2020-02-06 by DzikuVx

jerrynsr, 2021-06-09:
Would the AR Wing Pro presets in iNav 3.0 be a good starting point for the legacy AR Wing 900 ?

avsaase, 2021-06-09:
@jerrynsr there are no plane-specific presets anymore in 3.0. Instead there are two presets, one for plane with a tail and one for planes without, that should be good enough to get your plane in the air. From that point you do an autotune to get a tune that works for your plane.

jerrynsr, 2021-06-09:
Ok.  I used to use Autotune when the presets needed to be adjusted.  This is good to know.   I've got a few airplanes to build up for this season.  An Ar Wing Pro, Legacy AR Wing 900,  Zohd Dart XL,  Volantax Ranger amd A Volantax Ranger EX 757-3.  I found a low cost source of original Tau Labs Sparky 2.0 controllers.  I have one in my AR Mini Wing now, it flies extremely well in Manual, Angle, and Horizon mode.. this weekend I'm testing Auto Launch, RTH, and Loiter (with wings I call it "Holding Pattern").   Thanks for the reply.  I just get nervous when I hit the Autotune switch and wait for things to settle down, LOL.  Fly up to 60m and flick the Autotune switch with fingers crossed.

b14ckyy, 2021-06-09:
@jerrynsr the new default poresets are tested on all these planes and are stable. Only on the Ranger EX be careful and don't Fully trust RTH yet. Anything above 2m wingspan MUST be autotuned before using any autonomous modes. For smaller planes the generic presets should be stable. But always be careful before you have done the tune.