iNavFlight/inav-configurator #916: Can´t flash local file on Mac os (10.13.6)

Makeadrone, 2020-02-04:
Can´t flash (inav_2.3.0_SPRACINGAIRBIT.hex) to the Airbit board (F3 Evo based). Booted into DFU mode. Flashing works well from Betaflight, same hex file.  @hydra can confirm. 

OS: Mac os 10.13.6 (17G10021) (Macbook Pro 13)
Inav Configurator 2.4.0

hydra, 2020-02-04:
yes, I can confirm, I saw the behavior via teamviewer remote screen sharing session.  Also of note that it doesn't matter what hex file is flashed and @Makeadrone was also testing with the SPRACINGF3EVO target prior to the new target I built for the SPRACINGAIRBIT today.

* Configurator doesn't put the board into DFU mode if it's already running iNAV.
* Configurator won't flash the HEX if the board is manually placed in DFU mode.
* Latest BF configurator works fine on the same machine, same hex, same FC.

wx4cb, 2020-03-15:
same applies for windows 10 latest configurator (new install).

Cannot load local firmware. it pops up and asks for the file when you click open does nothing. loading online works fine however

PaulWieland, 2020-04-11:
This is still broken in 2.4.1. Have been using BetaFlight to flash all my iNav boards.

DzikuVx, 2020-05-19:
Tested with 10.15.4 and Configurator 2.5: works like expected

closed 2020-05-19 by DzikuVx