iNavFlight/inav-configurator #920: Update Dart 250G Preset

b14ckyy, 2020-02-07:
I have a Dart 250G PNP version and tried the preset from Configurator 2.4 with INAV 2.3 on the board. But this preset was unflyable. way too high PIDs and too low FF. Almost no pitch authority and nearly crashed. 

Also the default 50/50 Mix is too low throw. Even in manual mode it is not enough to safely control the plane and that was the main reason for missing pitch authority. If possible, the mixer need to be raised as the PNP servos have pretty small control angle. 

smix 0 3 0 80 0 -1
smix 1 3 1 70 0 -1
smix 2 4 0 -80 0 -1
smix 3 4 1 70 0 -1

set fw_p_pitch = 3
set fw_i_pitch = 7
set fw_ff_pitch = 40
set fw_p_roll = 2
set fw_i_roll = 4
set fw_ff_roll = 18

set gyro_hardware_lpf = 42HZ
set gyro_lpf_hz = 30

set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 360
set pitch_rate = 13

rsilk1949, 2020-06-11:
Has anyone else used and verified these settings?

b14ckyy, 2020-06-15:
about 20 people in the INAV FW Group and all are happy with it.

b14ckyy, 2020-06-15:
Update: I changed the mixer on pitch slightly as it was slightly too high and could cause nose jumping if the CG is critical.

rsilk1949, 2020-07-04:
Do these settings apply regardless of all up weight and battery power?
I am running 3S at 250gm weight.

rsilk1949, 2020-07-04:
I have found your interesting articles on inavfixedwinggroup.com
In the dart 250g beginners guide, the #tune section has different values for the pids, etc.
Pls clarify.

b14ckyy, 2020-07-04:
@rsilk1949 The Beginners Setup on the website is configured for high agility. The roll rate is ramped up to 520°/s and pitch rate to 140°/s. That's why the PIDs are different.

The Preset in configurator is just for initial flight to get it in the air stable and tune afterwards to the pilots needs. the preset rates are much slower for more gentle flying.

rsilk1949, 2020-07-04:
Tx for the clarification.
For a novice iNav flyer about to do their maiden flight, is it safe to take the beginners setup CLI settings file and replace the relevant lines from the preset notes (1st post above)?

b14ckyy, 2020-07-04:
yes you can do that. The Beginners setup includes a few more thing stuff like for auto navigation and Autolaunch. So You can take them over and take the settings from first post over then to have a more gentle flyer or a base for your own tune. 

And yes that should be 36. Thanks for the hint.

PR #1096 (2020-11-14): Update dart 250g preset# 920
Updated the preset in line with the suggested, proven settings. https://github.com/iNavFlight/inav-configurator/issues/920

I did not add the mixer settings, as I figured the preset is the wrong place for this. However, I added a note to the preset description, highlighting the required weights in the mixer.

Fixes #920
closed 2020-11-14 by DzikuVx