iNavFlight/inav-configurator #921: Don't spin motors does not work as expected

Jacky2k, 2020-02-09:
Hi everyone,
my drone just tried to kill me, although the "Don't spin the motors when armed" was turned on. I learned that this does only mean that the motors will not spin as long as you do not throttle up. But in my opinion that is not what the text says.
However, I suggest renaming that features and/or adding a text in the help box (it is currently empty) explaining this behavior.

Mateyhv, 2020-02-10:
A help box is always a good idea, however I am not sure I understand what the problem is. "Don't spin the motors when armed" does just that, you can arm without spinning motors. Once you throttle up the motors start spinning and only will stop on disarm. There is no point in stopping the motors midair since its the only way drones fly/controll/correct.

Jacky2k, 2020-02-10:
I interpreted this as "Don't spin the motors while armed". Which would actually be a nice feature when debugging. As I said, I learned that this means something different. So, I am happy if you put a help box next to it, which explains this in more detail.

PR #929 (2020-02-21): Extra infor on MOTOR_STOP feature in outputs tab
Close #921
closed 2020-02-21 by DzikuVx