iNavFlight/inav-configurator #924: Flashing problem Matek F405-STD using Linux

yajo10, 2020-02-18:
Hey,
I having problems with flashing a firmware newer than 1.7.3 to the FC Matek F405-STD. 
I'm using the latest inav-configurator 2.4.0 with linux (64). Flashing allways works and there is no error. But if I try to reconnect the board the blue LED will never blink for a long time and the board is not configurable. With the version 1.7.3 I can access the Configuration Menu using the port `/dev/ttyACM0` but with newer versions this port is not visible. 

So If I click disconnect with version 1.7.3 and try to flash a newer version without replugging the FC nor pressing the Boot-Button there will be the error: `Communication with bootloader failed`. If I reboot the device using the command  `dfu` flashing seem to work fine. (I don't have the 'No reboot sequenz' button selected)
The board reboots, the blue and red led is flickering some times but in the end the blue one is not blicking all the time. Just the red one is powered on a bit...
After that I can not connect to the configurations without reflashing to 1.7.3....

Does anybody have an Idea how to solve this?
Thanks in advance
yajo

yajo10, 2020-04-22:
The solutiion for me was to change my udev-Rules and change filter policies of ModemManager.
So I created them: `sudo nano /etc/udev/rules.d/50-flightcontrol.rules` and added the following content.
`# 0483:5740 - STM32F4 Dsicovery in USB Serial Mode (CN5)
ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", ENV{MTP_NO_PROBE}="1"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE:="0666"
KERNEL=="ttyACM*", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE:="0666"`

But this did not solved the problem already. Additionaly I had to change the filter policy of ModemManager according to this [solution](https://askubuntu.com/a/1119498) This worked for me. 

*But for flashing my fc I have to revert the changes!*

closed 2020-04-22 by yajo10

medisun, 2025-07-20:
For those who might have issues with flashing FC with error `Are udev rules installed correctly? See docs for instructions`
This works for me https://betaflight.com/docs/wiki/guides/current/Installing-Betaflight#platform-specific-linux

I am using arch linux and  inav configurator v8.1.

In my case i have to to do this:
```sh
sudo usermod -a -G uucp $USER

(echo '# DFU (Internal bootloader for STM32 and AT32 MCUs)'
	echo 'ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="2e3c", ATTRS{idProduct}=="df11", MODE="0664", GROUP="plugdev"'
	echo 'ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE="0664", GROUP="plugdev"') | sudo tee /etc/udev/rules.d/45-stdfu-permissions.rules > /dev/null
```