iNavFlight/inav-configurator #930: Unable to flash with Windows 10

rsilk1949, 2020-02-24:
I have just got a laptop with a fresh install of Windows 10.
Using Configurator 2.4.1, I am trying to install iNav 2.4.0 on a Matek411SE board.
When I push the flash button, the port dropdown does not change to DFU and the flash fails. It looks like the FC may have changed to flash mode as the red light is solid.
I looked at installing the STM driver, but the web site says it is no good for Win10.

DzikuVx, 2020-02-24:
1. Unplug the board
2. Press and hold boot button
3. Plug in USB
Board will go into DFU mode (you might need drivers)

closed 2020-02-24 by DzikuVx

rsilk1949, 2020-02-24:
I did above.
Solid red light on board but drop down does not change to "DFU".
Press Flash:
STM32 Communication failed, wrong response, expected: 121 (0x79) received: 0 (0x0)

rsilk1949, 2020-02-24:
FIXED: Watch this video to show how to reinstall the original Win 10 driver files with the new Microsoft drivers.
https://youtu.be/AeaglOYphfc