iNavFlight/inav-configurator #963: Allow re-ordering of modes page

MrD-RC, 2020-04-23:
Currently the modes page is a bit of a jumble, making it harder to see the set modes up; but even more so help other people with issues.

If the modes could be re-ordered by the user, that would be great. However, if not perhaps group the modes in a more user friendly way. For example:

Arm
Manual
Horizon
Angle
NAV RTH
NAV POSHOLD
NAV CRUISE
NAV ALTHOLD
NAV WP
etc.... 

The most used things closer to the top. Then grouped together appropriately; like flight modes, navigation, tuning, OSD, accessories (for example cam control, leds and beeper).

robertha64, 2020-04-24:
Good idea, even the re-arangement will be very good.
+1

wx4cb, 2020-05-28:
i mentioned a similar thing a couple years ago that was effectively ignored :D

https://github.com/iNavFlight/inav-configurator/issues/555

rsilk1949, 2020-11-15:
I would prefer the sort order to be by Channel #.
When I am checking for example a 3 way switch, it is often necessary to scroll the mode page up & down to check the PWM values are set as expected.

MrD-RC, 2020-11-15:
But surely all the modes on that switch will be the same type of mode?  So this would save scrolling

[image]

closed 2022-03-14 by DzikuVx