iNavFlight/inav-configurator #977: PID tab 2.4.0 broken on firmware 2.4.0

digitalentity, 2020-05-02:
Firmware: 2.4.0, airplane setup
PID tab doesn't load with this log in the console:

```
FC reports unsupported message error: 4103
script.js:22932 Unknown setting type null for setting 'dynamic_gyro_notch_enabled'
script.js:21432 FC reports unsupported message error: 4103
script.js:22932 Unknown setting type null for setting 'dynamic_gyro_notch_min_hz'
script.js:21432 FC reports unsupported message error: 4103
script.js:22932 Unknown setting type null for setting 'dynamic_gyro_notch_q'
script.js:18886 Unhandled rejection TypeError: Cannot read property '0' of undefined    at HTMLInputElement.<anonymous> (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:41651:46)    at Function.each (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:90:2882)    at n.fn.init.each (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:90:847)    at HTMLTableRowElement.<anonymous> (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:41650:33)    at Function.each (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:90:2882)    at n.fn.init.each (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:90:847)    at pid_and_rc_to_form (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:41644:39)    at process_html (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:41749:9)    at r (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:18888:7722)    at i._settlePromiseFromHandler (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:18887:13044)    at i._settlePromise (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:18887:13847)    at i._settlePromise0 (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:18887:14548)    at i._settlePromises (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:18887:15878)    at r._drainQueue (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:18886:3148)    at r._drainQueues (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:18886:3209)    at drainQueues (chrome-extension://modhfcbjjfkllpnmlfikolhikailmeno/build/script.js:18886:1238)
D @ script.js:18886
```

DzikuVx, 2020-05-02:
[image]

@digitalentity I don't think you have 2.4 on both sides.

DzikuVx, 2020-05-02:
That was Configurator 2.4.1 and Firmware 2.4

digitalentity, 2020-05-02:
Yes, Configurator 2.4.1 works, but don't we support previous version as
well?

DzikuVx, 2020-05-02:
Ah, so you tested master... OK, that changes it.

digitalentity, 2020-05-02:
Yeah, basically:

configurator `2.4.1`, firmware `2.4.0` - works
configurator `master`, firmware `2.4.0` - doesn't work
configurator `master`, firmware `development` - works

PR #978 (2020-05-02): Fix PID tab for 2.4 firmware
Fixes #977
closed 2020-05-02 by DzikuVx