iNavFlight/inav-configurator #979: Matek F722/F405/F765 wing stuck in FS mode

wx4cb, 2020-05-03:
titale says it all. I have to either zadig or "impulse driver fix" every time i plug in a flight controller.

Never did this before flashing with 2.4. I'm using the "SE" versions of the software and if i revert back to t.3 it all works fine The same issue as described here: https://github.com/iNavFlight/inav-configurator/issues/852

I have reflashed multiple times tryingto isolate this and it's still the same.
[image]
[image]

fiam, 2020-05-04:
@wx4cb You don't need to use zadig to change the drivers while in virtual com port mode. In fact, if you don't let windows use the default driver it's not gonna work. Zadig is only needed for changing the DFU driver. Follow these steps to uninstall the current driver https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it , then unplug the FC, plug it back again and let Windows install the drivers for the virtual com port itself.

wx4cb, 2020-05-04:
Thats the thing ive done all of that and it still does it. In fact, this screenshot is from a brand new install of win10 pro, o ly thing on there is the configurator. I checked it on there before i posted it

fiam, 2020-05-04:
@wx4cb With a clean install and without messing with 3rd party drivers, Windows 10 will install `usbser.sys` for a Virtual COM Port device. I tested it as recently as last week. As your screenshot shows, your system is using `WinUSB.sys`, which is intended for communicating with the device directly from user space i.e. it won't appear as a serial port until you make it use the correct driver. Are you sure you haven't installed any 3rd party drivers?

Something that might work is using Zadig to override the driver you have installed with `USB Serial (CDC)`. Use the small vertical arrows next to the "WinUSB..." until it looks like this, then press on Replace/Downgrade/Install driver (not sure what the big button will show in your case).

[image]

wx4cb, 2020-05-04:
> @wx4cb With a clean install and without messing with 3rd party drivers, Windows 10 will install `usbser.sys` for a Virtual COM Port device. I tested it as recently as last week. As your screenshot [...]

The only thing i did was to install the ST drivers so i could flash the board as it wouldn't go into DFU mode without it and then run the zadig like you've had to do for ages for that. other than that no i haven't.

I'[ll try removing all the devices (hidden too) and see what happens.

fiam, 2020-05-05:
@wx4cb You need to remove all STM32 devices (even the hidden ones) and make sure the option to delete the driver is checked, as detailed in https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it

Then let Windows install default drivers for VCP mode and use Zadig to force the DFU mode to use WinUSB.

closed 2021-06-26 by DzikuVx