iNavFlight/inav-configurator #980: Compass not calibrating

nosyninja, 2020-05-04:
## Current Behavior
INav Client fails to calibrate compass during the 30 second window. X Y Z all remain set to 0.
HMC5883 on SPRacing F3 Delux Flight Controller.

## Steps to Reproduce

1. Calibrate compass.
2. Save

## Expected behavior
X Y Z values set to numbers other than 0

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:

# version
# INAV/SPRACINGF3 2.2.1 Jul  3 2019 / 21:50:13 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

digitalentity, 2020-05-04:
Ok, so not a firmware issue. Firmware seems to finish the calibration and store the values, but Configurator fails to update the tab. @DzikuVx could you please have a look?

DzikuVx, 2020-05-04:
The workaround is as follows:

After calibration results in ZEROs in mag calibration:
1. Enter Setup tab
1. Go back to Calibration tab
1. Fields should be filled
1. Hit save

dimon777, 2020-05-11:
I had the same issue with MATEK F722-SE, blaming either my GPS/compass or FC. Since issue apparently is with the inav-configurator, can we post here a link to this bug?

closed 2020-05-12 by DzikuVx

emautner, 2020-08-09:
Matek F722-HD boards and Matek F722-Px boards only calibrate 1 time.. Subsequent attempts to calibrate using the 6-step accelerometer calibration result in a message stating that you've already calibrated.  The only way to calibrate is to completely start over and re-flash the FC.  The same bug exists in iNAV configurator 2.5.0 for both the optic flow sensor calibration, as well as the compass Calibration.  However, for both compass and optic flow calibration iNAV bugs, the bug doesn't reveal itself with the statement "you've already calibrated".. rather, it leads you to believe that calibration was performed.  However, the X,Y,Z and scale values never update.

PR #992 (2020-05-12): Compass calibration improvements
fixes #980