iNavFlight/inav-configurator #981: Mixer gui hows output incorectly

SchmidtDawid, 2020-05-04:
## Current Behavior
Mixer gui hows output incorectly. At screen below you can see that motor4 is on pad s5. But its not. It is on pad s6 cause Inav forcing ledstrip to s5 and dont even know it. 

[image]

## Steps to Reproduce
1. Have matek f405std
2. Try to add servo
3. Try to not break your servo
4. check resorce on cli

## Expected behavior
Expect to have possibility to set servo on s5 or s6 output

## Suggested solution(s)
resource remaping? - it is so "simple" solution for most issues there.

## Additional context
Lack of resource remaping i NOT user friendly - it is super user unfriendly when there is no way to fix the problem.

---
# INAV/MATEKF405 2.4.0 Feb 11 2020 / 22:43:00 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

DzikuVx, 2020-05-04:
this board has only one output that can be used as a Servo output in Multirotor configuration and it is not Configurator but hardware and firmware issue. Configurator correctly display output allocation in this case

closed 2020-05-04 by DzikuVx

SchmidtDawid, 2020-05-04:
I can setup at least 2 servos 4 motors an ledstrip on this board in Betaflight. So yes it is iNav issue.