iNavFlight/inav-configurator #985: Configurator 2.5.0 rc 1 - OSD Tab - Alarms - Temperature Labels

GregMichalski, 2020-05-10:
OSD Tab - Under Alarms there are 2 instances of Minimum Baro Temp. Presume second instance should actually be minimum ESC temp.
[image]

DzikuVx, 2020-05-12:
will be done in RC2

closed 2020-05-12 by DzikuVx

Jetrell, 2020-05-12:
@DzikuVx  When you say... done in RC2. Do you mean FW RC2? 
I would have thought Configurator RC2 would have fixed it. 
But it still displays `Minimum Baro temperature` alarm twice. The second should be` Minimum ESC temperature`.
 
[image]

Thanks

DzikuVx, 2020-05-13:
@giacomo892

DzikuVx, 2020-05-13:
In that case fixed in 36e6cf38b16a9bc3a6f50923f036c38376f00074 and will be released with final

Jetrell, 2020-05-13:
@giacomo892 @DzikuVx  Thanks guys :+1: