iNavFlight/inav-configurator #988: Map and Direction Arrow is not showing in configurator 2.5

NasirHayat, 2020-05-11:
I Found a bug in the new version of the configurator. As you move in Mission Control Tab There is a Map and arrow sign (which show heading of dron on map) it is really useful feature when you going to set the orientation of the compass. This the map and heading arrow was working fine with older version 2.4.1 but not working in 2.5. I think this the best way to double-check compass orientation and calibration is fine if this mission control options works fine. 
[image]

DzikuVx, 2020-06-11:
[image]

Works like expected

closed 2020-06-11 by DzikuVx

tourerjim, 2020-06-25:
> I Found a bug in the new version of the configurator. As you move in Mission Control Tab There is a Map and arrow sign (which show heading of dron on map) it is really useful feature when you going [...]

Mines same on two quads I've setup now and can't work it out, get the Map GPS tab but not Missions tab

tourerjim, 2020-06-27:
Solved, you have to keep zooming out